Maker-Management-Platform/agent #59: stls not being inserted

C0untZero, 2024-01-17:
I'm running mmp through docker-compose on my rpi5.
I put a bunch of stls in various folders in my mmp library and the projects are not auto created.
When I check out the agent logs I see just the first stl with the message produced png, rendering png,
and then just endless spam of the logs for printer connectivity.

Am I doing something wrong?

javijuji, 2024-01-17:
I am having the same issue. Library isn't getting scanned.

EduardoOliveira, 2024-01-18:
Hi
@C0untZero  does it stop on the first model? Does the image get written into the library?

@javijuji Are you also running on a RPI5?

C0untZero, 2024-01-19:
Hi hi,

Yeah, it does the first image and then nothing happens.

EduardoOliveira, 2024-01-19:
Hi could you please test this branch https://github.com/Maker-Management-Platform/agent/pkgs/container/agent/168820804?tag=missing-error-handling-for-rendering ? I've made some changes that might solve it

C0untZero, 2024-01-24:
Same behavior on that branch as well.

EduardoOliveira, 2024-01-24:
Could you post logs for discovery?

C0untZero, 2024-01-25:
Not much in the logs
[mmp.txt](https://github.com/Maker-Management-Platform/agent/files/14050394/mmp.txt)

EduardoOliveira, 2024-01-26:
I'm gonna try to reproduce the issue on an RPI, I will only have access to a RPI4...
I've downloaded a couple of Rumble minies to test... they are pretty detailed... compared to other models I have, Jana takes about 3 times to render... Might the ARM be a bottleneck there?
Whats the CPU load? I would assume you would get 1 core to 100% and drop after the image is rendered, the render finish is not in your logs...
Can you describe the CPU behavior when you boot MMP?

C0untZero, 2024-01-26:
Yeah, all 4 cores actually get smashed to 100%. I've tried leaving it on like that for a while but no luck.
[image]

You can test on the actual stls I'm using, they're available here - https://warcraftrumble.blizzard.com/en-us/collection.
This is the first one on the list which stops everything - https://assets.warcraftrumble.blizzard.com/blueprints/warcraft_rumble_alliance_jainaproudmoore.zip.

EduardoOliveira, 2024-02-15:
Hi, sorry for the late response...
I went and bought a rpi5 8gb.
I was able to reproduce the problem with some models.
"Normal models" work just fine, a little slower than on my server, but high res models like the ones from Rumble take way too long.

Looking at the library that I'm currently using to render the models, It does a CPU count and uses all the cores to compute the render... Not configurable...
I'll think of something as a workaround while I try to contact the library developer or test with others.

Worst case scenario, rendering detaches from discovery (it should so it doesn't block the whole usage of the app), each model goes into render with a configurable timeout, if it reaches the timeout goes into some kind of blacklist.

C0untZero, 2024-02-17:
Wow, that's commitment to a project :D Getting an rpi5 just to reproduce a bug.
Thanks man! 

Yeah, I guess it would makes sense to split discovery and rendering, I'd be fine with not having some models on preview as long as I can list them in the library at least.

[image]

EduardoOliveira, 2024-02-19:
[image]
Running on an RPI5 8gb :)
I hope to have something you can run soon

closed 2024-02-22 by EduardoOliveira