Maker-Management-Platform/agent #60: panic: runtime error: invalid memory address or nil pointer dereference

Fabaderheld, 2024-01-18:
Hi,

during the initial discovery process in one folder I receive this error
```panic: runtime error: invalid memory address or nil pointer dereference```
It always happens in the same folder but it does not look suspicious

here is the content of it
```
cap-back-left.stl                 main-left-front.stl                tpu-cap-front-left.stl
cap-back-left.stl.render.png      main-left-front.stl.render.png     tpu-cap-front-left.stl.render.png
cap-back-right.stl                main-right-back.stl                tpu-cap-front-right.stl
cap-back-right.stl.render.png     main-right-back.stl.render.png     tpu-cap-front-right.stl.render.png
cap-front-left.stl                main-right-front.stl               tpu-left-side-1.stl
cap-front-left.stl.render.png     main-right-front.stl.render.png    tpu-left-side-1.stl.render.png
cap-front-right.stl               peg-b.stl                          tpu-left-side-2.stl
cap-front-right.stl.render.png    peg-b.stl.render.png               tpu-left-side-2.stl.render.png
cap-left-side-2.stl               peg-f.stl                          tpu-right-side-1.stl
cap-left-side-2.stl.render.png    peg-f.stl.render.png               tpu-right-side-1.stl.render.png
cap-right-side-2.stl              peg-l.stl                          tpu-right-side-2.stl
cap-right-side-2.stl.render.png   peg-l.stl.render.png               tpu-right-side-2.stl.render.png
caps-left-side-1.stl              peg-r.stl                          tpu-top-bl.stl
caps-left-side-1.stl.render.png   peg-r.stl.render.png               tpu-top-bl.stl.render.png
caps-right-side-1.stl             tpu-back-left.stl                  tpu-top-br.stl
caps-right-side-1.stl.render.png  tpu-back-left.stl.render.png       tpu-top-br.stl.render.png
magnet-holder.stl                 tpu-back-right.stl                 tpu-top-fl.stl
magnet-holder.stl.render.png      tpu-back-right.stl.render.png      tpu-top-fl.stl.render.png
main-left-back.stl                tpu-cablehole-plug.stl             tpu-top-fr.stl
main-left-back.stl.render.png     tpu-cablehole-plug.stl.render.png  tpu-top-fr.stl.render.png
``` 

any idea why this is happening?

Thanks

Faba

EduardoOliveira, 2024-01-18:
Hi
Is that the root of your library?
MMP currently assumes projects are folders inside the library mount point, to be honest I haven't tested with files in the root

Fabaderheld, 2024-01-19:
Hi,

no they are in a subfolder

```/library/@uncollected/dk-riser#59```

other files in ```/library/@uncollected``` work fine

If I can test anything just let me know

regards

Faba

EduardoOliveira, 2024-01-19:
Could you try breaking that into multiple projects with one asset, so we could pinpoint what asset is the culprit and go from there?

Fabaderheld, 2024-01-19:
Sure I'll do that an get back to you

EduardoOliveira, 2024-01-19:
I've made some improvements in the area of the code in this branch, can you test it out?
https://github.com/Maker-Management-Platform/agent/pkgs/container/agent/168820804?tag=missing-error-handling-for-rendering

It was failing hard if the STL wasn't completely valid, now it skips it

Fabaderheld, 2024-01-19:
Will do!

Progress I made so far. I just moved the folder out of the /library and the error kept coming. So I removed the whole ```@uncollected``` since I suspected the ```@``` to be an issue. The good news, the error disappeared but know at some point the container just restarts.

Fabaderheld, 2024-01-21:
Quick update, 

the container crashed were caused by running out of memory, I increased the memory and it didn't happen anymore.

I added all files back to a folder without the @ and it still crashes, I'll try to figure out which file is causing it.

closed 2024-02-15 by EduardoOliveira