Maker-Management-Platform/agent #70: 3mf file not refreshed

Fabaderheld, 2024-01-21:
Hi,

I've been playing around with it and I don't know if this is intend, a bug or maybe something to improve.

When I made changes to a 3mf file and delete all the generated preview, the only way to force a refresh is to restart the agent container. And then the interesting part is, that all gets regenerated but from the previous version of the 3mf file, not the one currently residing in the filesystem.

regards

Faba

EduardoOliveira, 2024-01-22:
Hi

yes, currently the 3fm data is extracted on upload or on discovery (server start), but, it's generated from the 3mf (zip) that's present in the filesystem, it shouldn't be possible for MMP to extract data from a previous version of the file.
Can you detail the issue a bit more?

Fabaderheld, 2024-01-23:
I'll try :)

1. I have a project containing a 3mf files and maker management generated .pngs for all the plates.
2. Then I opened the 3mf from the project folder and removed a plate from it.  
3. Trying to force regeneration of the .pngs by deleting all the files and restarted the container
4. To my surprise the files generated are the same as in step 1

EduardoOliveira, 2024-01-23:
Thank you, I'll take a look at it